ccutrer/waterfurnace_aurora #4: Trying to get the monitor working

madasus, 2021-08-28:
Am I calling this right? I have an IP to RS485 adapter which I can telnet to on port 8234 to see the traffic on the serial side of the device.  I'm running ruby 2.7.4 on a docker and calling aurora_montor with the uri in the command line.  I tried tcp:// also.  Do I need to do something else to get it working directly over the network? it seemed like I didn't need to use ser2net based on the instructions but maybe I am mistaken.

root@ruby1:/# aurora_monitor telnet://IP:8234/
Traceback (most recent call last):
	7: from /usr/local/bundle/bin/aurora_monitor:23:in `<main>'
	6: from /usr/local/bundle/bin/aurora_monitor:23:in `load'
	5: from /usr/local/bundle/gems/waterfurnace_aurora-0.3.5/exe/aurora_monitor:40:in `<top (required)>'
	4: from /usr/local/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:83:in `require'
	3: from /usr/local/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:83:in `require'
	2: from /usr/local/bundle/gems/net-telnet-rfc2217-0.0.5/lib/net/telnet/rfc2217.rb:2:in `<top (required)>'
	1: from /usr/local/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:83:in `require'
/usr/local/lib/ruby/2.7.0/rubygems/core_ext/kernel_require.rb:83:in `require': cannot load such file -- byebug (LoadError)

ccutrer, 2021-08-28:
Sigh. I probably have a stray debugger statement left in. You can probably work around by doing `gem install byebug` first. But I'm not positive that your adapter is speaking RFC2217. Telnet is more than happy to let you talk to anything without supporting telnet escape codes. If it's not, you should change to using tcp://ip:8234/.

madasus, 2021-08-28:
perfect - the error is gone now - thank you.  No data when I use telnet and an error when i use TCP://

Traceback (most recent call last):
	5: from /usr/local/bundle/bin/aurora_monitor:23:in `<main>'
	4: from /usr/local/bundle/bin/aurora_monitor:23:in `load'
	3: from /usr/local/bundle/gems/waterfurnace_aurora-0.3.5/exe/aurora_monitor:46:in `<top (required)>'
	2: from /usr/local/bundle/gems/waterfurnace_aurora-0.3.5/exe/aurora_monitor:46:in `new'
	1: from /usr/local/bundle/gems/ccutrer-serialport-1.0.0/lib/ccutrer/serial_port.rb:7:in `initialize'
/usr/local/bundle/gems/ccutrer-serialport-1.0.0/lib/ccutrer/serial_port.rb:7:in `initialize': Is a directory @ rb_sysopen - / (Errno::EISDIR)

I think the adapter supports RFC2217 so telnet should do the trick - possible I have a wiring issue so i'll have to return to the unit with my laptop tomorrow

ccutrer, 2021-08-28:
I'll also double check the tcp:// functionality. The error you posted shows it was trying to open a local serial port.

ccutrer, 2021-08-28:
oh lol because I never added support for it. new release coming up shortly...

madasus, 2021-08-28:
Don’t rush on my account.  I will recheck my cable tomorrow and see make sure it really works

ccutrer, 2021-08-28:
0.3.6 is on rubygems now with tcp:// support. and fixing the byebug thing. easy fixes

madasus, 2021-08-29:
Thanks - confirm both fixed.

Reviewed my cable and it is good.  If I use my breakout box to tap the RS485 bus between the AWL and the machine I see traffic.  When I plugin to the AID tool port on the AWL I don't see anything (using monitor or fetch).  Any thoughts?

ccutrer, 2021-08-29:
when plugged into the aid tool port on the AWL, you shouldn't see anything with monitor - there's no other traffic to see. monitor is passive only - reporting what traffic it sees on the bus. fetch requires an additional argument of which register to fetch. I just published 0.3.7 of the gem to ruby gems that gives it a full syntax, and documents it: https://github.com/ccutrer/waterfurnace_aurora#aurora_fetch. But really at this point it's easiest if you can start using the MQTT Bridge.

madasus, 2021-08-29:
thank you - still striking out here.  it must be my ip-->rs485 box.  I'm using Ruby in a docker so using a USB device is a little more tricky than an ip device.  let me re-trace my steps and i'll just build another cable temporarily.  Will report back later!

ccutrer, 2021-08-29:
👍  I'm a little surprised it's not just working. If aurora_monitor is spitting out valid data, it means the serial parameters are correct. aurora_fetch and aurora_mqtt_bridge (assuming you switch your connection point, but keep the cable the same configuration) should use the same connection params, but will need to write. Only thing I can think of is the IP bridge isn't flushing the writes or something (on either the ruby side or the IP-to-serial bridge side).

ccutrer, 2021-08-29:
if you want more synchronous help, shoot me an email and I'll send you an invite to a Slack instance where we can chat. there are a couple other people in there at the moment also getting it set up.

ccutrer, 2021-09-02:
all issues have been cleared up in Slack

closed 2021-09-02 by ccutrer