ccutrer/waterfurnace_aurora #9: Kudos

kskenyon, 2022-01-19:
Nice work on reverse engineering.  I have two WF and this has been on my list of things to look into for a while.  Great way to take some load off the symphony.mywaterfurnace.com servers.  I have been using a python script to pull the data from their server.  They don't like it if the hits are too frequent.  MQTT was a great choice.

K

ccutrer, 2022-01-30:
Thank you! Glad more and more people are able to utilize it! Makes me feel like all the hours I sat on a folding chair in my mechanical room with a laptop on my lap wired in trying to decipher things are having more value. Or more correctly - the time I probably should have spent with my wife and kids ;).

closed 2022-01-30 by ccutrer

kskenyon, 2022-03-04:
I finally got to try aurora_mqtt_bridge for real and I can't get anything through serial.  I'm using an auto switching RS485 to serial adapter in a Pi.   I had it working for a while with a 18B20 to 485 adapter so I think everything should work.  I get this error when running aurora_monitor

pi@pi3B-plus:/etc/systemd/system $ sudo aurora_monitor /dev/ttyS0
Traceback (most recent call last):
	6: from /usr/local/bin/aurora_monitor:23:in `<main>'
	5: from /usr/local/bin/aurora_monitor:23:in `load'
	4: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-1.4.2/exe/aurora_monitor:61:in `<top (required)>'
	3: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-1.4.2/exe/aurora_monitor:61:in `new'
	2: from /var/lib/gems/2.5.0/gems/ccutrer-serialport-1.1.0/lib/ccutrer/serial_port.rb:28:in `initialize'
	1: from /var/lib/gems/2.5.0/gems/ccutrer-serialport-1.1.0/lib/ccutrer/serial_port.rb:44:in `configure'
/var/lib/gems/2.5.0/gems/ccutrer-serialport-1.1.0/lib/ccutrer/serial_port.rb:117:in `apply': Invalid argument (Errno::EINVAL)

sudo didn't make any difference.
Any ideas?

Kevin

kskenyon, 2022-03-04:
Mar 03 23:53:09 pi3B-plus systemd[1]: Started Aurora MQTT Bridge.
Mar 03 23:53:11 pi3B-plus aurora_mqtt_bridge[24307]: /var/lib/gems/2.5.0/gems/ccutrer-serialport-1.1.0/lib/ccutrer/serial_port.rb:117:in `apply': Invalid argument (Errno::EINVAL)
Mar 03 23:53:11 pi3B-plus aurora_mqtt_bridge[24307]:         from /var/lib/gems/2.5.0/gems/ccutrer-serialport-1.1.0/lib/ccutrer/serial_port.rb:44:in `configure'
Mar 03 23:53:11 pi3B-plus aurora_mqtt_bridge[24307]:         from /var/lib/gems/2.5.0/gems/ccutrer-serialport-1.1.0/lib/ccutrer/serial_port.rb:28:in `initialize'
Mar 03 23:53:11 pi3B-plus aurora_mqtt_bridge[24307]:         from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-1.4.2/lib/aurora/abc_client.rb:41:in `new'
Mar 03 23:53:11 pi3B-plus aurora_mqtt_bridge[24307]:         from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-1.4.2/lib/aurora/abc_client.rb:41:in `open_modbus_slave'
Mar 03 23:53:11 pi3B-plus aurora_mqtt_bridge[24307]:         from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-1.4.2/lib/aurora/abc_client.rb:153:in `initialize'
Mar 03 23:53:11 pi3B-plus aurora_mqtt_bridge[24307]:         from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-1.4.2/exe/aurora_mqtt_bridge:42:in `new'
Mar 03 23:53:11 pi3B-plus aurora_mqtt_bridge[24307]:         from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-1.4.2/exe/aurora_mqtt_bridge:42:in `<top (required)>'
Mar 03 23:53:11 pi3B-plus aurora_mqtt_bridge[24307]:         from /usr/local/bin/aurora_mqtt_bridge:23:in `load'
Mar 03 23:53:11 pi3B-plus aurora_mqtt_bridge[24307]:         from /usr/local/bin/aurora_mqtt_bridge:23:in `<main>'
Mar 03 23:53:11 pi3B-plus systemd[1]: aurora_mqtt_bridge.service: Main process exited, code=exited, status=1/FAILURE
Mar 03 23:53:11 pi3B-plus systemd[1]: aurora_mqtt_bridge.service: Failed with result 'exit-code'.
Mar 03 23:53:14 pi3B-plus systemd[1]: aurora_mqtt_bridge.service: Service RestartSec=3s expired, scheduling restart.
Mar 03 23:53:14 pi3B-plus systemd[1]: aurora_mqtt_bridge.service: Scheduled restart job, restart counter is at 278.
Mar 03 23:53:14 pi3B-plus systemd[1]: Stopped Aurora MQTT Bridge

ccutrer, 2022-03-04:
The EINVAL implies your serial adapter is refusing to accept the serial port parameters (baud, parity, etc.) that WaterFurnace requires. By "auto switching" adapter, are you referring to a MAX485 that needs an additional pin to switch between send and receive? Or the opposite- your adapter automatically detects when it's doing a send?

kskenyon, 2022-03-04:
It auto switches between send and receive.
[image]
[image]

I also have an ethernet adapter but I couldn't figure out how to set that up either.  tcp:// and telnet:// URI'S didn't work.  The instructions are auto translated from Chinese.  Not easy to understand. 

[image]

ccutrer, 2022-03-04:
https://forums.raspberrypi.com/viewtopic.php?t=198999 looks relevant. WaterFurnace uses even parity. So maybe try switching the UARTs in your pi's boot config?

kskenyon, 2022-03-04:
I think I got serial working OK now on ttyAMA0 with your pointer.  But I still get this error when running aurora_mqtt_bridge and nothing when I run aurora_monitor.  Maybe it's just a bum converter.  I have another on order so I'll have to wait and try again.

/var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:275:in `rescue in query': Timed out during read attempt (ModBus::Errors::ModBusTimeout)

ccutrer, 2022-03-04:
Did you try switching +/- and TX/RX? You probably got it right, but sometimes cheap electronics are mislabeled. I love me some cheap electronics, but they do have occasional pitfalls

kskenyon, 2022-03-06:
Finally got it going with the red adapter.  Not sure why it started working.  Polyfuse blown?

Nice data flowing to MQTT now.  

I repeat, Kudos for developing this app.

kskenyon, 2022-03-06:
Can I run two of your app with different serial ports to monitor two WaterFurnaces from one Pi?  Can I just make a duplicate of the app with a different name and run a separate systemd? Or just run two systemd with the same app? I don't suppose many people have two waterfurnaces in one house.  One is for forced air and filtration, the other is for underfloor hydronic heating and domestic hot water.  They both work great.  The forced air pulls warm air from the high ceiling over the woodstove in winter and cool air from the finished walkout basement in the summer.  Almost free cooling and very affordable heating.

ccutrer, 2022-03-07:
Yes, there are multiple people running two instances. Just make another systemd unit with a different name, and change its configuration to pass the proper serial port path. In MQTT, they will show up with slightly different paths based on their serial numbers.