ccutrer/waterfurnace_aurora #10: Illegal Data Address error

doctorfred, 2022-01-26:
Thank you for creating this!

My system is a Waterfurnace Series 7, IntelliZone 2, DHW, VS Drive, VS Pump

I built a cable as documented, using the DSD TECH SH-U11 adapter.

When starting Aurora MQTT bridge, I see an error:

Jan 26 16:10:52 raspberrypi aurora_mqtt_bridge[1276]: /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:283:in `query': The data address received in the query is not an allowable address for the server (ModBus::Errors::IllegalDataAddress)
Jan 26 16:10:52 raspberrypi aurora_mqtt_bridge[1276]:         from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:174:in `read_holding_registers'
Jan 26 16:10:52 raspberrypi aurora_mqtt_bridge[1276]:         from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:178:in `read_holding_register'
Jan 26 16:10:52 raspberrypi aurora_mqtt_bridge[1276]:         from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/proxy.rb:13:in `[]'
Jan 26 16:10:52 raspberrypi aurora_mqtt_bridge[1276]:         from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.5/lib/aurora/modbus/slave.rb:26:in `[]'
Jan 26 16:10:52 raspberrypi aurora_mqtt_bridge[1276]:         from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.5/lib/aurora/abc_client.rb:78:in `initialize'
Jan 26 16:10:52 raspberrypi aurora_mqtt_bridge[1276]:         from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.5/exe/aurora_mqtt_bridge:38:in `new'
Jan 26 16:10:52 raspberrypi aurora_mqtt_bridge[1276]:         from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.5/exe/aurora_mqtt_bridge:38:in `<top (required)>'
Jan 26 16:10:52 raspberrypi aurora_mqtt_bridge[1276]:         from /usr/local/bin/aurora_mqtt_bridge:23:in `load'
Jan 26 16:10:52 raspberrypi aurora_mqtt_bridge[1276]:         from /usr/local/bin/aurora_mqtt_bridge:23:in `<main>'
Jan 26 16:10:52 raspberrypi systemd[1]: aurora_mqtt_bridge.service: Main process exited, code=exited, status=1/FAILURE

Is this something specific to my system and firmware?  I get a different error if I unplug the cable from the ABC.

ccutrer, 2022-01-27:
Hmm... that's almost my exact configuration, so I'm surprised you would get that error. Can you do `aurora_fetch /dev/ttyUSB0 valid --yaml > myheatpump.yml` (adjusting the serial port if necessary) and then e-mail me the resulting file (assuming it doesn't run into errors)?

doctorfred, 2022-01-27:
Same thing: 

aurora_fetch /dev/ttyUSB0 valid --yaml
Traceback (most recent call last):
	9: from /usr/local/bin/aurora_fetch:23:in `<main>'
	8: from /usr/local/bin/aurora_fetch:23:in `load'
	7: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.5/exe/aurora_fetch:35:in `<top (required)>'
	6: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.5/exe/aurora_fetch:35:in `new'
	5: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.5/lib/aurora/abc_client.rb:78:in `initialize'
	4: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.5/lib/aurora/modbus/slave.rb:26:in `[]'
	3: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/proxy.rb:13:in `[]'
	2: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:178:in `read_holding_register'
	1: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:174:in `read_holding_registers'
/var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:283:in `query': The data address received in the query is not an allowable address for the server (ModBus::Errors::IllegalDataAddress)

I’m wondering if there’s something wrong with the port or bit rate or some such.  I don’t know much of anything about RS-485/RS-422.

ccutrer, 2022-01-27:
oh right, that makes sense... it goes through the same initialization first. lemme look and see if I can simplify that code to get past that part, and get to the part where it tries everything it can looking for valid registers

ccutrer, 2022-01-27:
okay, 0.7.6 released that should bypass your error on initialization, and then try _really hard_ to get all the data it can.

doctorfred, 2022-01-27:
This time it looks like it’s hanging on the read:

Successfully installed waterfurnace_aurora-0.7.6
2 gems installed
***@***.***:/usr/local/bin $ aurora_fetch /dev/ttyUSB0 valid
^CTraceback (most recent call last):
	25: from /usr/local/bin/aurora_fetch:23:in `<main>'
	24: from /usr/local/bin/aurora_fetch:23:in `load'
	23: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/exe/aurora_fetch:49:in `<top (required)>'
	22: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:64:in `query_registers'
	21: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:64:in `each'
	20: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:65:in `block in query_registers'
	19: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:71:in `rescue in block in query_registers'
	18: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:71:in `each'
	17: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:72:in `block (2 levels) in query_registers'
	16: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:77:in `rescue in block (2 levels) in query_registers'
	15: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:77:in `each'
	14: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:78:in `block (3 levels) in query_registers'
	13: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/modbus/slave.rb:26:in `[]'
	12: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/proxy.rb:13:in `[]'
	11: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:178:in `read_holding_register'
	10: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:174:in `read_holding_registers'
	 9: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:267:in `query'
	 8: from /usr/lib/ruby/2.5.0/timeout.rb:103:in `timeout'
	 7: from /usr/lib/ruby/2.5.0/timeout.rb:93:in `block in timeout'
	 6: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:269:in `block in query'
	 5: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/rtu_slave.rb:32:in `read_pdu'
	 4: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/modbus/slave.rb:35:in `read_rtu_response'
	 3: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/rtu.rb:43:in `read'
	 2: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/rtu.rb:43:in `loop'
	 1: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/rtu.rb:47:in `block in read'
/var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/rtu.rb:47:in `wait_readable': Interrupt

I hit ctrl-c after awhile.

ccutrer, 2022-01-27:
It might take awhile. Possibly several minutes. You can also add `--no-try-individual` and it won't try _quite_ so hard, but might be missing some data.

doctorfred, 2022-01-27:
I can see the send/receive light on the rs422 dogle going nuts and the send/receive on the ABC main board flashing i respose to your aurora_fetch.  I’ll wait awhile.

Thank you for helping!

doctorfred, 2022-01-27:
Ok - it quit by itself:

***@***.***:/usr/local/bin $ aurora_fetch /dev/ttyUSB0 valid
Traceback (most recent call last):
	10: from /usr/local/bin/aurora_fetch:23:in `<main>'
	 9: from /usr/local/bin/aurora_fetch:23:in `load'
	 8: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/exe/aurora_fetch:49:in `<top (required)>'
	 7: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:64:in `query_registers'
	 6: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:64:in `each'
	 5: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:65:in `block in query_registers'
	 4: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:71:in `rescue in block in query_registers'
	 3: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:71:in `each'
	 2: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:72:in `block (2 levels) in query_registers'
	 1: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/modbus/slave.rb:14:in `read_multiple_holding_registers'
/var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:283:in `query': The function code received in the query is not an allowable action for the server (ModBus::Errors::IllegalFunction)

and

***@***.***:/usr/local/bin $ aurora_fetch /dev/ttyUSB0 valid --yaml --no-try-individual
Traceback (most recent call last):
	10: from /usr/local/bin/aurora_fetch:23:in `<main>'
	 9: from /usr/local/bin/aurora_fetch:23:in `load'
	 8: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/exe/aurora_fetch:49:in `<top (required)>'
	 7: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:64:in `query_registers'
	 6: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:64:in `each'
	 5: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:65:in `block in query_registers'
	 4: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:71:in `rescue in block in query_registers'
	 3: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:71:in `each'
	 2: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/abc_client.rb:72:in `block (2 levels) in query_registers'
	 1: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.6/lib/aurora/modbus/slave.rb:14:in `read_multiple_holding_registers'
/var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:283:in `query': The function code received in the query is not an allowable action for the server (ModBus::Errors::IllegalFunction)

was much faster, but same output.

ccutrer, 2022-01-27:
0.7.7 will ignore that error as well and keep going.

doctorfred, 2022-01-27:
—no-try-individual:

---
0: 0
1: 0
2: 204
3: 201
4: 32767
5: 240
6: 0
7: 0
8: 0
9: 120
10: 120
11: 0
12: 1
13: 0
14: 30
15: 30
16: 238
17: 400
18: 0
19: 662
20: 646
21: 1023
22: 0
23: 829
24: 1
25: 0
26: 32775
27: 589
28: 409
29: 15
30: 584
31: 400
32: 32767
33: 254
34: 60
35: 1
36: 2
37: 0
38: 0
39: 0
40: 0
41: 0
42: 65535
43: 0
44: 0
45: 0
46: 65535
47: 0
48: 0
49: 0
50: 2
51: 3
52: 10
53: 32767
54: 2
55: 65535
56: 65535
57: 65535
58: 65535
59: 65535
60: 65535
61: 65535
62: 65535
63: 65535
64: 65535
65: 65535
66: 65535
[... 788 lines cut ...]
3914: 144
60050: 5900
60051: 5854
60052: 10230
60053: 8273
60100: 0
60101: 4
60102: 3
60103: 4
60104: 362
60105: 0
60106: 4
60107: 243
60108: 14584
60109: 35370
61000: 123
61001: 14424
61002: 36288
61003: 0
61004: 0
61005: 0
61006: 0
61007: 0
61008: 0
61009: 0

doctorfred, 2022-01-27:
And w/o no-try-individual, it errors:

***@***.***:/usr/local/bin $ aurora_fetch /dev/ttyUSB0 valid --yaml > ~/out.yml
Traceback (most recent call last):
	17: from /usr/local/bin/aurora_fetch:23:in `<main>'
	16: from /usr/local/bin/aurora_fetch:23:in `load'
	15: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.7/exe/aurora_fetch:49:in `<top (required)>'
	14: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.7/lib/aurora/abc_client.rb:64:in `query_registers'
	13: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.7/lib/aurora/abc_client.rb:64:in `each'
	12: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.7/lib/aurora/abc_client.rb:65:in `block in query_registers'
	11: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.7/lib/aurora/abc_client.rb:71:in `rescue in block in query_registers'
	10: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.7/lib/aurora/abc_client.rb:71:in `each'
	 9: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.7/lib/aurora/abc_client.rb:72:in `block (2 levels) in query_registers'
	 8: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.7/lib/aurora/abc_client.rb:77:in `rescue in block (2 levels) in query_registers'
	 7: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.7/lib/aurora/abc_client.rb:77:in `each'
	 6: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.7/lib/aurora/abc_client.rb:78:in `block (3 levels) in query_registers'
	 5: from /var/lib/gems/2.5.0/gems/waterfurnace_aurora-0.7.7/lib/aurora/modbus/slave.rb:26:in `[]'
	 4: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/proxy.rb:13:in `[]'
	 3: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:178:in `read_holding_register'
	 2: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:174:in `read_holding_registers'
	 1: from /var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:266:in `query'
/var/lib/gems/2.5.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:275:in `rescue in query': Timed out during read attempt (ModBus::Errors::ModBusTimeout)

ccutrer, 2022-01-27:
:nod: I'll try and look at the data tomorrow, see if I can make sense of it, and request more if I need to.

doctorfred, 2022-01-27:
Problem was my system was at Firmware 2.0.4. Fixed in 0.8.0.  Thanks!

closed 2022-01-27 by doctorfred