ccutrer/waterfurnace_aurora #13: Limits of each implementation

ballpython06, 2022-01-30:
Hello,

will the first code listed and plugging into the awl’s aid port collect the data that the furnace sends to symphony every 10 seconds? I am looking to access the data the dealers receive on their portal, but I want to record the historical data. Will this work or do I need to create the connection between the abc and awl? Also, is the Ethernet cable that comes with the awl use the standard t568b? Thank you for this awesome code.

ccutrer, 2022-01-30:
I'm not sure what you're referring to with "first code listed", so I'll try to explain more.

The general way to use this software is to have the aurora_mqtt_bridge constantly running as a system and publishing messages to MQTT. From there, you can have a home automation system (like Home Assistant or OpenHAB) consume the MQTT messages and allow you to see state, control things, and implicitly record history. I'm pretty sure there's also plugins for Telegraf/InfluxDB to directly consume MQTT messages and store the history in InfluxDB. If you're wanting to store the history in some other way, and looking to bypass MQTT, that should be possible, but you'll have to write your own ruby code that uses this gem directly, and then outputs the data in whatever way you want.

As to if this is the same data as is sent to Symphony and the dealer portal - yes and no. This tool has access to the exact same data that an AWL (and thus Symphony) can access, it's a matter of _finding_ the data you want. I've mapped almost every register that an AID tool or the AWL/Symphony Web UI can display for my system, so I know where to find the data. But I've never seen the dealer portal on Symphony, so cannot say with certainty that I'm pulling every piece of information that has. You can look through my list of register names at https://github.com/ccutrer/waterfurnace_aurora/blob/096f930b962e37193e89144e7a6395de3f266eaa/lib/aurora/registers.rb#L697 to see if the data you're hoping for is mapped already (beware that naming might be slightly different - this is true even of WaterFurnace's tools - the AID tool and Symphony might have slightly different naming for the same data point). The MQTT bridge defaults to refreshing the data every 5 seconds while it's running.

As for where to connect: if you have an AWL, and you want to use an AWL _and_ this tool, you _must_ connect to the AWL's AID Tool port. If you no longer need/want an AWL, or don't have one, you can connect directly to the WaterFurnace (actually the ABC board is what you talk to). Either way, the same data is available - an AWL just forwards any requests on its AID tool port to the ABC. You _cannot_ use the MQTT bridge _and_ an AWL if you tap in between the ABC and the AWL - the messages will clobber each other and neither tool will work.

As to the wiring of the AWL-provided cable: I'm not sure, but if I remember correctly the provided cable doesn't even use the standard 4 colors. IIRC it is a "straight" cable, and not a "crossover", but you'll have to identify wires by their pin location in one of the plugs, rather than by color.

ballpython06, 2022-01-30:
Thank you for the response. So if I plug into the aid tool port on the awl, it will pull data every seconds?

ccutrer, 2022-01-30:
Yup

closed 2022-01-31 by ccutrer

ccutrer, 2022-02-07:
Btw, I had a technician over today, and he confirmed that what he sees on the Symphony website is exactly the same as I see with the Web AID Tool locally, and the ruby gem is currently publishing (almost?) everything accessible in the AID Tool.

ballpython06, 2022-02-08:
I have had it running for a few days now and the outside temp stopped
updating, but the symphony app still updates. Any idea on how to fix that
one data point? Thank you.

ccutrer, 2022-02-08:
To be clear, the aurora-xxx/abc/outdoor-temperature topic in MQTT has stopped updated, but other data points _are_ updating, like aurora-xx/abc/entering-air-temperature? That would be quite unusual for a single data point to stop working, since all data is fetched from the ABC in a batch. It would make more sense to me if somehow the bridge had either stopped running, or was jammed up, and _no_ data was updating anymore.

ccutrer, 2022-02-08:
Hmm. Do you have an IZ2 system, or just a single communicating thermostat? The register I'm pulling the data from looks like it might be in an IZ2-only range, so if you don't have IZ2 I might need to find where the register is for non-IZ2.

If you restart the aurora_mqtt_bridge, does it start updating again? If it doesn't, try publishing the message `31003` to `aurora-xxx/$modbus` (replacing xxx as appropriate with your serial number), and then it should send back a message at `aurora-xxx/$modbus/31003` with a fresh reading. Is that reading still "stuck"?

ballpython06, 2022-02-08:
I’m sorry, but I am still new at some of this. How exactly do I send a
message on ssh? Can you give me the exact command please? I have the serial
number, I just don’t know how to start the command to publish a message.

ccutrer, 2022-02-08:
It's MQTT. I would recommend using MQTT Explorer (http://mqtt-explorer.com) as a simple UI to see what data is in your MQTT server.

ballpython06, 2022-02-09:
Thank you. I sent the message and now the mqtt explorer shows no temp and
home assistant is showing the reading as zero. I then uninstalled the gem
and mosquito and then reinstalled them. Now it is again Stu k at 12
degrees.