ccutrer/waterfurnace_aurora #14: Active Fault handeling ?

Cogginshome, 2022-01-30:
2 things 

first this is a wonder piece of software 
and I can add "Water furnace Versatec Base" to the compatible list.
mine don't have a lot of the options but for what I was looking for it's working well 

Now my questions
is there a way to send a clear fault history to the board over mqtt or on the pi running the bridge ?

is there a easy way to tell if there is an active fault?
I see the history counters and there is a lockout on the current mode but I would like to have home assistant try and prevent a lock out if possible

ccutrer, 2022-01-30:
Yes. Send "clear" to homie/<your device>/faults/clear-history/set, and it will reset the fault history count to 0 for all faults. It is _not_ the same as "reset lockout" from your thermostat, which I cannot do.

And no, it doesn't look like I'm exposing an active fault to MQTT. I have the information, and I had already thought of doing this yesterday when I noticed that my thermostat was giving an E5, but was _not_ locked out. I'll get it added within a day or two.

ccutrer, 2022-01-30:
I've never even heard of the Versatec series. Glad it's working! I'd be interested in getting a dump of all of the registers in your system (using `aurora_fetch /dev/ttyUSB0 valid --yaml > registers.yml` as documented in the readme), if you're willing. Might help me identify some things available on your system that aren't on others. You can e-mail me the resulting file (don't post it directly to GitHub, since it will have your system's serial number in it).

Cogginshome, 2022-01-31:
can't get aurora_fetch /dev/ttyUSB0 valid to run I can pull batches of registers I hit problems in the 150-200 range some place

pi@raspberrypi:~ $ aurora_fetch /dev/ttyUSB0 valid
Traceback (most recent call last):
	13: from /usr/local/bin/aurora_fetch:23:in `<main>'
	12: from /usr/local/bin/aurora_fetch:23:in `load'
	11: from /var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.0.1/exe/aurora_fetch:49:in `<top (required)>'
	10: from /var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.0.1/lib/aurora/abc_client.rb:65:in `query_registers'
	 9: from /var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.0.1/lib/aurora/abc_client.rb:65:in `each'
	 8: from /var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.0.1/lib/aurora/abc_client.rb:66:in `block in query_registers'
	 7: from /var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.0.1/lib/aurora/modbus/slave.rb:12:in `read_multiple_holding_registers'
	 6: from /var/lib/gems/2.7.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:267:in `query'
	 5: from /var/lib/gems/2.7.0/gems/timeout-0.2.0/lib/timeout.rb:117:in `timeout'
	 4: from /var/lib/gems/2.7.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:269:in `block in query'
	 3: from /var/lib/gems/2.7.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/rtu_slave.rb:42:in `read_pdu'
	 2: from /var/lib/gems/2.7.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/rtu_slave.rb:42:in `loop'
	 1: from /var/lib/gems/2.7.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/rtu_slave.rb:44:in `block in read_pdu'
/var/lib/gems/2.7.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/rtu_slave.rb:44:in `sleep': execution expired (ModBus::Errors::ModBusTimeout)
	8: from /usr/local/bin/aurora_fetch:23:in `<main>'
	7: from /usr/local/bin/aurora_fetch:23:in `load'
	6: from /var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.0.1/exe/aurora_fetch:49:in `<top (required)>'
	5: from /var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.0.1/lib/aurora/abc_client.rb:65:in `query_registers'
	4: from /var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.0.1/lib/aurora/abc_client.rb:65:in `each'
	3: from /var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.0.1/lib/aurora/abc_client.rb:66:in `block in query_registers'
	2: from /var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.0.1/lib/aurora/modbus/slave.rb:12:in `read_multiple_holding_registers'
	1: from /var/lib/gems/2.7.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:266:in `query'
/var/lib/gems/2.7.0/gems/rmodbus-ccutrer-2.1.0/lib/rmodbus/client/slave.rb:275:in `rescue in query': Timed out during read attempt (ModBus::Errors::ModBusTimeout)

ccutrer, 2022-01-31:
1.1.0 just released should be more fault tolerant when fetching lots of data. You can also try `known` and/or `all`. all > valid > known in terms of how many registers it tries, but `all` might actually be more fault tolerant because it will try to fetch in even blocks instead of jumping around.

closed 2022-01-31 by ccutrer

Cogginshome, 2022-02-19:
Did you get my file?

ccutrer, 2022-02-28:
I did. I haven't had time to go through it looking for differences yet, but I have used it to verify the data is present in your system (and everyone else's) before adding a new feature.

relevante, 2024-01-20:
> Yes. Send "clear" to homie//faults/clear-history/set, and it will reset the fault history count to 0 for all faults. It is _not_ the same as "reset lockout" from your thermostat, which I cannot do.

Is there a specific reason that "reset lockout" isn't possible from the gem, or has it just not been reverse engineered yet?

My system is designed to run without antifreeze for extra efficiency, but when it's very cold outside, it sometimes trips the freeze detection and locks out. I'd like to be able to catch the fault and perhaps bump the thermostat down a degree or two and then reset the lockout. I'll likely put some effort into figuring out the lockout reset unless there's a known reason it's not feasible. Thanks.

ccutrer, 2024-01-21:
I've not found any evidence that a lockout can be reset from the "front" Modbus bus, only from the thermostat bus which is a separate protocol and has not been reverse engineered. You _can_ reset lockout by using the separate Emergency Shutdown input on the board (it may have to be configured to be such an input with an AID Tool), which I use a smart relay to control.

relevante, 2024-01-21:
My installer can reset it remotely via his web interface. That command should be able to be captured and replicated with your tools, right?

relevante, 2024-01-21:
Actually, I have one locked out right now and I went into my Symphony app to make sure there wasn’t anything there. There isn’t directly, but I just turned it off and back to heat mode and it started running. So maybe that’s all it takes.