ccutrer/waterfurnace_aurora #18: Negative subcool-temperature, missing heating-liquid-line-temperature

svennest, 2022-03-06:
First of all, thanks so much for all of your work on this, really nice documentation and install recipe!

I dedicated an old Raspberry Pi model B for this running Raspbian 10 (buster) with the following adapter (the adapter you linked was out of stock at the time) and it worked right away. Uses only 3% CPU on average and very little memory. The Pi gets wired network and power from a POE adapter at my furnace, no need for an outlet and I have the POE switch on a UPS.
https://www.amazon.com/gp/product/B081MB6PN2

A couple of minor items I noticed:

/compressor/subcool-temperature (register 1135) is publishing a negative value. The register value from aurora_fetch is negative, the number is correct based on register 1134 - register 1109 (Subcool = High Pressure Saturation Temp - Liquid Line Temperature) , just the wrong sign?

/compressor/heating-liquid-line-temperature is missing, I see it listed as register 1109, can this be added to the published values?

svennest, 2022-03-09:
Thanks for adding the LLT value. I don't have an AID tool, does yours display a negative value for SubCool?

From section 9 of page 57 of the 700A11 installation manual (it says 49 on the PDF page looks like they missed renumbering) :
https://www.waterfurnace.com/literature/7series/im2700ana.pdf 

SubCool = High Pressure Saturation Temp - Liquid Line Temperature

High Pressure Saturation Temp is /compressor/saturated-condensor-discharge-temperature

The LLT sensor is shown on the diagram on page 55, it is the temperature after exiting the air coil. Nice heating cycle diagram on page 56 too. It should be lower than the discharge temperature and a bit higher than the leaving air temperature.

In the Subcooling column of Operating Parameters on page 53, all of the subcooling values are positive.

ccutrer, 2022-03-09:
This is a view of my system as of a minute ago. [image]

I've had some issues with my system lately and had a tech out twice. He definitely looked at the SubCool intently, and was not phased at all by it being negative.

svennest, 2022-03-17:
Very interesting, well if the AID tool shows it as negative then best to leave it that way. Maybe both of our systems have similar issues. I'll try to remember to ask my tech about the negative number next time he is here.

ccutrer, 2022-05-16:
I'm closing for now. Feel free to re-open if you find additional evidence.

closed 2022-05-16 by ccutrer