ccutrer/waterfurnace_aurora #19: zones not being picked up by node-red-contrib-node-homie-red v0.2.0

kskenyon, 2022-03-09:
For some reason the above node-red node picks up everything but the individual zones.  I have three and they show up OK in MQTT.  Could there be some formatting issue in the homie MQTT that prevents the zone info from showing up in the pull down in node-red property?

[image]

ccutrer, 2022-03-09:
I don't see any issues with the Homie metadata. OpenHAB's Homie implementation also reads everything correctly. Is there any Node-RED logging that might offer a clue?

kskenyon, 2022-03-09:
Nothing in the Node-RED debug log.  The Property pull-down menu just doesn't populate for any of the zones.  Everything else seems to work as you can see.  I don't have OpenHAB or Homeassistant setup to check those.  I'll try that next.

kskenyon, 2022-03-18:
Home Assistant seems to work OK.  Not sure what the problem in Node-Red is caused by.

ccutrer, 2023-09-17:
Closing as it doesn't appear to be a problem with this gem.

closed 2023-09-17 by ccutrer