ccutrer/waterfurnace_aurora #24: Control or only monitoring?

relevante, 2022-11-12:
Hello, this looks awesome and once I get my system (7 series) installed, hopefully within the next couple of months, I'm sure I'll set it up to monitor it within HA.

Curious though if there has been any work done in terms of actually taking over low level control of the compressor and blower requests? In reading about the Intellizone logic, it seems like there's room for improvement and I'm curious what the jumping off point would be for trying to intervene. From a protocol standpoint do you have any idea how difficult that might be? Like I said, I don't even have my system yet, so I'm still pretty much in the dark on a lot of aspects. And hopefully it'll be fine out of the box, but would like to have a backup plan just in case I'm not thrilled with the control. Thanks!

ccutrer, 2022-11-12:
Yes. On a 7 series there are settings for directly controlling compressor and blower speeds. I can't remember if I set them up to work via MQTT - I was meaning to. So if/when you get to that point, and if it doesn't work, reopen this issue and I'll flesh it out.

closed 2022-11-12 by ccutrer

relevante, 2025-01-23:
@ccutrer Wondering if you've had a chance to look into this since this thread? If you do, I have a few custom control ideas I would like to try out on my system. I took a quick look at the code trying to figure out how to implement it, but it didn't immediately jump out at me. Thanks!