ccutrer/waterfurnace_aurora #27: Little help troubleshooting the MQTT part

bpsmicro, 2023-01-05:
I'm not super-familiar with linux/etc (I'm a Windows guy). I have this running on a rPi A+. I can see the blinky lights on the RS485 converter that tells me it's getting data from the WF. When I SSH into the rPi, all looks "normal" to me. I followed the instructions to the letter.

Using MQTT Explorer, I plug in the IP of the rPi (to avoid DNS issues), leave the two switched (Validate certificate & Encryption) OFF. I've tried with the username/password I set up the rPi with, and also blank. When I click Connect, I get nothing but "Disconnected from server" over and over.  Nothing informative.

If I turn ON both the switched, I get an ECONNREFUSED error instead, which implies it's *trying* to talk to the broker.

I'm also trying to connect from my Home Assistant box, but details on exactly how to accomplish that are unclear, and I'm getting even less useful error information.

Any thoughts on how to troubleshoot this?

bpsmicro, 2023-01-05:
Aha! Stumbled across an article at https://randomnerdtutorials.com/how-to-install-mosquitto-broker-on-raspberry-pi/, and at the very bottom there's a segment "Mosquitto Broker Enable Remote Access (No Authentication)". Followed the instructions, and MQTT Explorer magically started to work.

I seem to be making progress in Home Assistant too. Leaving this here in case somebody else runs into a similar problem.

jairusmartin, 2023-01-07:
Run `mosquitto -v`  in one terminal and in the second terminal run `aurora_mqtt_bridge /dev/ttyUSB0 mqtt://localhost`

To enable any connections (no security) for troubleshooting, add the following to you mosquitto.conf: 

```
allow_anonymous true
listener 1883
persistence true
persistence_location /mosquitto/data/
```

I am working on a single command setup using docker compose, here is my initial config if your interested:
```
version: '3.3'

services:
  mqtt-broker:
    container_name: mqtt-broker
    image: eclipse-mosquitto
    restart: unless-stopped
    ports:
      - 1883:1883
      # - 8883:8883 # TLS
    # healthcheck:
    #     test: ["CMD", "curl", "-f", "http://localhost:15672"]
    #     interval: 30s
    #     timeout: 10s
    #     retries: 5
    volumes:
      - mosquitto-config:/mosquitto/config
      - mosquitto-data:/mosquitto/data
      - mosquitto-log:/mosquitto/log
    networks:
      - wf-net

  nodered:
    # container_name: nodered
    image: nodered/node-red:latest
    restart: unless-stopped
    depends_on:
      - mqtt-broker
     #  - mqtt-bridge
    environment:
      - TZ=America/New_York
    ports:
      - 1880:1880
    networks:
      - wf-net
    volumes:
      - nodered-data:/data

  bridge:
    container_name: mqtt-bridge
    build: 
    #  context: https://github.com/ccutrer/waterfurnace_aurora.git#main
    #  dockerfile: ./docker/Dockerfile
      context: ./
      dockerfile: Dockerfile
    image: ccutrer/waterfurnace_aurora
    restart: unless-stopped
    command: "aurora_mqtt_bridge /dev/ttyUSB0 mqtt://mqtt-broker"
    depends_on: 
      - mqtt-broker
      - nodered
    devices:
      - '/dev/ttyUSB0'
    environment:
      - TTY=/dev/ttyUSB0
      - MQTT=mqtt://mqtt-broker
    networks:
      - wf-net

networks:
  wf-net:

volumes:
  nodered-data:
  mosquitto-config:
  mosquitto-data:
  mosquitto-log:
```

jeremyamoore, 2023-07-31:
For Home Assistant, I installed the Mosquito Addon in HomeAssistat and then pointed the MQTT URL in /etc/systemd/system/aurora_mqtt_bridge.service to that instead of the localhost.  You will have to add username and password to the URL.  You should then see the "homie" and "homeassistant" topics there and all the new entities in Home Assistant.  This would mean you do not need the Mosquito server running on the WaterFurnace bridge host.

closed 2023-08-21 by bpsmicro