ccutrer/waterfurnace_aurora #37: line-voltage is tied to Line Voltage Setting?

svennest, 2023-04-03:
/abc/line-voltage never changes and is tied to register 112 (Line Voltage Setting).

It would be helpful to expose the current line voltage reading from register 16 (Line Voltage).

What do you think about changing line-voltage to register 16, or to expose both publishing 16 to line-voltage and 112 to line-voltage-setting?

ccutrer, 2023-07-29:
There is no line voltage sensor, so it will never change. You can _set_ the voltage (normally via an AID tool) via MQTT, and that’s how wattage is calculated for the energy monitors (the sensors technically only measure amperage).

svennest, 2023-07-31:
In this project doc/registers.md says: |16||Line Voltage|

On my WaterFurnace 7, register 112 reads a constant value of 244.

Register 16 fluctuates (239 at the moment) and matches the value displayed on the Energy Monitor screen of a physical AID tool as Line Voltage, also verified with a meter at the unit.

commit 6b0c1404 (2023-09-17): expose detected line voltage

fixes #37
closed 2023-09-17 by ccutrer