ccutrer/waterfurnace_aurora #39: Monitoring multiple systems on same host

jeremyamoore, 2023-07-31:
I have one unit working and feeding HomeAssistant (connected to AWL port) but would like to know if I can monitor multiple units on the same host with multiple RS485 to USB adapters.  I see the MQTT side has the serial number there so should not overwrite things.  Does a separate instance of the bridge need to be installed, or would additional line in the config for /dev/ttyUSB1 work (maybe that is a feature request)?

ccutrer, 2023-07-31:
You would need to run two instances of the service, each pointed at different serial ports.  Then yes, the rest of it works fine, not stomping on each others data.

jeremyamoore, 2023-08-01:
Thanks, that works for me!

I just did a:

`cp /etc/systemd/system/aurora_mqtt_bridge.service /etc/systemd/system/aurora_mqtt_bridge_USB1.service`

Then edited the file so it references USB1 rather than USB0 and did the regular:

`sudo systemctl enable aurora_mqtt_bridge_USB1`
`sudo systemctl start aurora_mqtt_bridge_USB1`

Both services are pointing to my Home Assistant Mosquitto add-on install.  I am not sure that the USB0/1 will stay associated to the same hardware port on reboot, but does not really matter as they are feeding the same MQTT endpoint.

FYI - I am using the following for USB ports and power to a Raspberry Pi Zero 2 W and seems to be doing OK.  Am not using the ethernet on there, just WiFi:
https://www.amazon.com/gp/product/B00L32UUJK

closed 2023-08-01 by jeremyamoore