ccutrer/waterfurnace_aurora #40: Not seeing leaving air temp and some other sensors

jeremyamoore, 2023-07-31:
I am not seeing the leaving air temp topic listed.  In the code I see where it is only adding that if awl_communicating. I assume that is looking for a communicating thermostat?  I am using my Series 5 with an EcoBee but do have this value available .

Looks like some of the pressure sensors are only pulled if the system is variable speed, but can also be available if they have the "Refrigerant Monitoring" kit installed on a serial 5.  So they could be broken out from the VS check.

ccutrer, 2023-09-17:
Can you email me a dump of your registers made with `aurora_fetch /dev/ttyUSB0 valid --yaml > myheatpump.yml`?

jeremyamoore, 2023-11-14:
Sorry about that, just seeing your request.  File should be attached.
[myheatpump.yml.zip](https://github.com/ccutrer/waterfurnace_aurora/files/13344604/myheatpump.yml.zip)

closed 2026-02-10 by ccutrer