ccutrer/waterfurnace_aurora #41: It appears that the serial driver is not being installed in the standard gem installation

phdweather, 2023-08-01:
Here is the error:

sudo /usr/local/bin/aurora_mqtt_bridge /dev/ttyUSB0 mqtt://localhost/
[WARNING] Install `ccutrer-serialport` gem for use RTU protocols

I had this running fine for about six months before my PI died. I replaced it and using the same steps as before, there is no data being published on the MQTT Explorer now because there is no working driver for the USB to serial installed according to the error code above. Maybe it got dropped off in the update 2 months ago.

Let me know what you think.

Thanks for all the work on this software. It is the best.

ccutrer, 2023-08-01:
What's the output of `gem list`? I ask, because `ccutrer-serialport` is clearly a runtime dependency of the latest release: https://rubygems.org/gems/waterfurnace_aurora/versions/1.4.6

phdweather, 2023-08-01:
Happy to run a fresh install again and capture that for you.

Thanks for the reply.  Again, thank you for developing this gem. It truly
means a lot and is awesome!

Here is the fresh output. It does look like it is in there. Not sure why
the warning for the driver when executing the bridge service. The error was
hidden and I had to force it to show the execution output to see why no
traffic was flowing.

sudo gem install rake waterfurnace_aurora --no-doc --verbose
HEAD https://index.rubygems.org/
200 OK
GET https://index.rubygems.org/info/rake
200 OK
/var/lib/gems/2.7.0/gems/rake-13.0.6/History.rdoc
/var/lib/gems/2.7.0/gems/rake-13.0.6/MIT-LICENSE
/var/lib/gems/2.7.0/gems/rake-13.0.6/README.rdoc
/var/lib/gems/2.7.0/gems/rake-13.0.6/doc/command_line_usage.rdoc
/var/lib/gems/2.7.0/gems/rake-13.0.6/doc/example/Rakefile1
/var/lib/gems/2.7.0/gems/rake-13.0.6/doc/example/Rakefile2
/var/lib/gems/2.7.0/gems/rake-13.0.6/doc/example/a.c
/var/lib/gems/2.7.0/gems/rake-13.0.6/doc/example/b.c
/var/lib/gems/2.7.0/gems/rake-13.0.6/doc/example/main.c
/var/lib/gems/2.7.0/gems/rake-13.0.6/doc/glossary.rdoc
/var/lib/gems/2.7.0/gems/rake-13.0.6/doc/jamis.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/doc/proto_rake.rdoc
/var/lib/gems/2.7.0/gems/rake-13.0.6/doc/rake.1
/var/lib/gems/2.7.0/gems/rake-13.0.6/doc/rakefile.rdoc
/var/lib/gems/2.7.0/gems/rake-13.0.6/doc/rational.rdoc
/var/lib/gems/2.7.0/gems/rake-13.0.6/exe/rake
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/application.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/backtrace.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/clean.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/cloneable.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/cpu_counter.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/default_loader.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/dsl_definition.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/early_time.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/ext/core.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/ext/string.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/file_creation_task.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/file_list.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/file_task.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/file_utils.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/file_utils_ext.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/invocation_chain.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/invocation_exception_mixin.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/late_time.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/linked_list.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/loaders/makefile.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/multi_task.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/name_space.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/packagetask.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/phony.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/private_reader.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/promise.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/pseudo_status.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/rake_module.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/rake_test_loader.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/rule_recursion_overflow_error.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/scope.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/task.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/task_argument_error.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/task_arguments.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/task_manager.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/tasklib.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/testtask.rb
/var/lib/gems/2.7.0/gems/rake-13.0.6/lib/rake/thread_history_display.rb
[... 60 lines cut ...]
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/aux_heat.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/blower.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/component.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/compressor.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/core_ext/string.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/dhw.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/humidistat.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/iz2_zone.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/mock_abc.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/modbus/server.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/modbus/slave.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/mqtt_modbus.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/pump.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/registers.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/thermostat.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/version.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/aurora/web_aid_tool.rb
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/lib/waterfurnace_aurora.rb
/usr/local/bin/aurora_monitor
/usr/local/bin/aurora_mqtt_bridge
/usr/local/bin/web_aid_tool
/usr/local/bin/aurora_mock
/usr/local/bin/aurora_fetch
Successfully installed waterfurnace_aurora-1.4.6
2 gems installed

ccutrer, 2023-08-01:
Hmm... but you're not getting any other errors? That warning doesn't necessarily mean it failed to talk to the serial port. It just means when the rmodbus gem loaded, the serial port gem wasn't yet available (since it's a soft dependency, this is allowed). But it should give an error later and immediately fail with something like `uninitialized constant CCutrer (NameError)` if it wasn't available when it actually tried to use it. When you run the bridge, does it just hang with no output? Does it give a stack trace when you ctrl-C it?

phdweather, 2023-08-01:
No errors otherwise. It just has no data coming in under the default #
subscription. I see the 192.168.0.148 and no output like normal.

I am going to replace the serial usb adapter next. This issue all
happened during a storm a couple weeks ago. I still have a solid red light
on it and the standard boot up lights when the PI boots up, but no other
activity or flashing on that status light. Since it corrupted my SD card
with the Raspberry Pi on it, I am going to assume maybe some ESD damage as
well. I have already tried a new Raspberry Pi 4 (the old one was still
working after cutting a new SD card with a new image on it), but the last
thing to try is the new usb adapter. I will get back to you when I have
that swapped out. I was just curious about the error as I have never tried
executing that before with the verbose mode to see how the loading was
going. It may be like you state in that the error has always been hiding in
the background and is normal.

Hopefully it is not the AWL!!!! Such a beautiful and easy thing to hook up
to the AID tool port of the AWL!

ccutrer, 2023-08-01:
Oh you’re going through an AWL? First try cutting that out of the loop. AWLs are known to die (mine has been dead for quite some time), and bypassing it would be much easier to test than waiting for a fresh USB serial adapter. You can also use the aurora_fetch command to test basic communication with your heat pump without any complexity of connecting to MQTT.

phdweather, 2023-08-02:
The AWL is working great for sending the data to the normal WF Symphony
website and I can see the basic data flowing still. But I agree with your
point as if it was an ESD event, it probably fried both ends (PI/adapter
and the AWL AID port). I should have the replacement serial usb adapter by
NLT tomorrow and it is only a 20 second swap out. So after exhausting that
part, I will just use a RJ45 straight through coupler to go from the PI to
the WF directly and bypass that AWL.

phdweather, 2023-08-03:
I tried the new USB adapter and it is still not working. I think you are
correct in that the AWL may have had that AID port go out on it. Currently
the other port on the AWL is hooked into my WF Sereies 7 and data is
flowing onto the WF website. Should I just use a straight-through RJ45
coupler to connect to the RJ45 coming from the USB adapter to bypass the
AWL? Since it is a straight-through coupler, the pins should match on both
sides.

Thanks for your help.

phdweather, 2023-08-03:
Here are the fetch and dump outputs. It appears there is still a load error
preventing the aurora bridge from accepting commands. It executes normally
with the start command and has no errors or outputs. But when trying to
execute the the fetch commands directly, we get a better look at the error
that is hidden. Looks like this is the error...    cannot load such file --
2.7/ffi_c (LoadError)

***@***.***:~ $ aurora_fetch /dev/ttyHeatPump 745-765
[WARNING] Install `ccutrer-serialport` gem for use RTU protocols
Traceback (most recent call last):
10: from /usr/local/bin/aurora_fetch:23:in `<main>'
9: from /usr/local/bin/aurora_fetch:23:in `load'
8: from
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/exe/aurora_fetch:5:in
`<top (required)>'
7: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
6: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
5: from
/var/lib/gems/2.7.0/gems/ccutrer-serialport-1.1.0/lib/ccutrer-serialport.rb:4:in
`<top (required)>'
4: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
3: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
2: from /var/lib/gems/2.7.0/gems/ffi-1.15.5/lib/ffi.rb:3:in `<top
(required)>'
1: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require': cannot load such file -- 2.7/ffi_c (LoadError)
11: from /usr/local/bin/aurora_fetch:23:in `<main>'
10: from /usr/local/bin/aurora_fetch:23:in `load'
9: from
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/exe/aurora_fetch:5:in
`<top (required)>'
8: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
7: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
6: from
/var/lib/gems/2.7.0/gems/ccutrer-serialport-1.1.0/lib/ccutrer-serialport.rb:4:in
`<top (required)>'
5: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
4: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
3: from /var/lib/gems/2.7.0/gems/ffi-1.15.5/lib/ffi.rb:2:in `<top
(required)>'
2: from /var/lib/gems/2.7.0/gems/ffi-1.15.5/lib/ffi.rb:5:in `rescue in <top
(required)>'
1: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require': /var/lib/gems/2.7.0/gems/ffi-1.15.5/lib/ffi_c.so: undefined
symbol: ffi_prep_cif_machdep -
/var/lib/gems/2.7.0/gems/ffi-1.15.5/lib/ffi_c.so (LoadError)
***@***.***:~ $ aurora_monitor /dev/ttyHeatPump -q
[WARNING] Install `ccutrer-serialport` gem for use RTU protocols
Traceback (most recent call last):
10: from /usr/local/bin/aurora_monitor:23:in `<main>'
9: from /usr/local/bin/aurora_monitor:23:in `load'
8: from
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/exe/aurora_monitor:60:in
`<top (required)>'
7: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
6: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
5: from
/var/lib/gems/2.7.0/gems/ccutrer-serialport-1.1.0/lib/ccutrer-serialport.rb:4:in
`<top (required)>'
4: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
3: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
2: from /var/lib/gems/2.7.0/gems/ffi-1.15.5/lib/ffi.rb:3:in `<top
(required)>'
1: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require': cannot load such file -- 2.7/ffi_c (LoadError)
11: from /usr/local/bin/aurora_monitor:23:in `<main>'
10: from /usr/local/bin/aurora_monitor:23:in `load'
9: from
/var/lib/gems/2.7.0/gems/waterfurnace_aurora-1.4.6/exe/aurora_monitor:60:in
`<top (required)>'
8: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
7: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
6: from
/var/lib/gems/2.7.0/gems/ccutrer-serialport-1.1.0/lib/ccutrer-serialport.rb:4:in
`<top (required)>'
5: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
4: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
3: from /var/lib/gems/2.7.0/gems/ffi-1.15.5/lib/ffi.rb:2:in `<top
(required)>'
2: from /var/lib/gems/2.7.0/gems/ffi-1.15.5/lib/ffi.rb:5:in `rescue in <top
(required)>'
1: from /usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require'
/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require': /var/lib/gems/2.7.0/gems/ffi-1.15.5/lib/ffi_c.so: undefined
symbol: ffi_prep_cif_machdep -
/var/lib/gems/2.7.0/gems/ffi-1.15.5/lib/ffi_c.so (LoadError)

ccutrer, 2023-08-03:
> I tried the new USB adapter and it is still not working. I think you are correct in that the AWL may have had that AID port go out on it. Currently the other port on the AWL is hooked into my WF [...]

Yes, you just unplug from the AID tool port on the AWL, unplug the AWL from the WF, then plug the USB adapter into the AID tool port on the WF. If your setup is such that that's not convenient, you can use a straight through double-RJ45 jack coupler to bypass the AWL.

> Looks like this is the error...    cannot load such file --
2.7/ffi_c (LoadError)

Whoa, that's... unexpected. That would point to either an issue with the FFI gem, or a problem with the Ruby installation. https://github.com/ffi/ffi/issues/1016 looks like the underlying issue. So try he suggested workaround: `gem install ffi -- --enable-system-libffi`.

phdweather, 2023-08-03:
> FFI gem, or a problem with the Ruby installation. ffi/ffi#1016

***@***.***:~ $ sudo gem install ffi -- -- enable-system-libffi
Building native extensions with: '-- enable-system-libffi'
This could take a while...
Successfully installed ffi-1.15.5
Parsing documentation for ffi-1.15.5
Installing ri documentation for ffi-1.15.5
Done installing documentation for ffi after 18 seconds
1 gem installed

I then stopped the bridge, disabled the bridge, enabled the bridge, and
started the bridge again. Same exact error when using the fetch command as
before. I figured it was at the ruby level based on the error code. I
checked out the path deep inside ruby as well and agree with your analysis.
Possibly something not compiling correctly as the syntax seems to be
correct. I have wiped the PI install a few times and repeated your setup
and been able to duplicate this ruby error. The error is only visible when
fetching or trying to get a dump. Obviously the MQTT is not able to get any
output either with the fetch not working. So I will stick with the fetch
command for now while we are troubleshooting.

phdweather, 2023-08-08:
I still get the same errors using the fetch command on the terminal window
of the PI computer even with the cable plugged straight into the WF AID
port.

/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb:85:in
`require': /var/lib/gems/2.7.0/gems/ffi-1.15.5/lib/ffi_c.so: undefined
symbol: ffi_prep_cif_machdep -
/var/lib/gems/2.7.0/gems/ffi-1.15.5/lib/ffi_c.so (LoadError)

Should I just wait until the ffi guy updates his software for the 2.7.0
version to get it working again? The $ sudo gem install ffi -- --
enable-system-libffi did not seem to have any effect on the error.

ccutrer, 2023-08-08:
You should post in the FFI issue saying that you're still having problems, even using the system libffi.

phdweather, 2023-08-14:
Sounds good. In the meantime, is there any way to get an older version
prior to 2.7.0? Maybe the older ffi will run fine in that version. I did
not see how to get an older version.

ccutrer, 2024-10-07:
I believe newer Raspberry Pi OS and Ruby version should work now.

closed 2024-12-28 by ccutrer

phdweather, 2024-12-28:
> Latest Ruby/Raspberry Pid OS as well as docker images no longer have an
> FFI issue.

Awesome!

Thanks Cody. I will do a fresh reinstall of everything and let you know!

phdweather, 2025-02-10:
So....

Fresh install of Pi. I believe it was an October 2024 update for the RPi4
64 bit.  Went through the basic stuff:

sudo apt install ruby ruby-dev
sudo gem install rake waterfurnace_aurora --no-doc
sudo apt install mosquitto
sudo curl https://github.com/ccutrer/waterfurnace_aurora/raw/main/contrib/aurora_mqtt_bridge.service
-L -o /etc/systemd/system/aurora_mqtt_bridge.service
sudo nano /etc/systemd/system/aurora_mqtt_bridge.service # edit the
service user to match your user
sudo systemctl enable aurora_mqtt_bridge
sudo systemctl start aurora_mqtt_bridge

In addition, did the normal steps for mosquitto config after that and added
listener 1883 and allow anonymous true lines to allow the MQTT to connect.
It connects both with raspberry.local and the internal IP address, but
there is no subscription traffic messages even with just # coming though.

Anything you want me to paste for output again?

phdweather, 2025-02-10:
Oh yeah....

Most importantly, I now own my own WF AID tool and I have verified that it
works via the gateway and also when directly connected to the WF Port. So
the serial bus is working correctly.