ccutrer/waterfurnace_aurora #44: MQTT Updates

starsoccer, 2023-08-23:
How often should data published via mqtt be updated, or is data only updated when an actual change takes place? Asking as I see some data that gets updated every few seconds, and then other points that seem to get 2 or 3 updates, and then never update again.

A good example of this is cooling set point. It gets updated when started, and then never updates again.

ccutrer, 2023-08-23:
Any particular value is only published to MQTT when it changes (and at startup, to set the initial state). All messages are sent retained, so even if a subscriber wasn't around when it was originally sent, they'll still get the most recent value.

starsoccer, 2023-08-23:
Well right after posting I was able to resolve this. For anyone else who hits this under the query options you can choose a fill option and select previous for it to just backfill the value

closed 2023-08-23 by starsoccer