ccutrer/waterfurnace_aurora #45: Multiple Devices on a Single Machine

starsoccer, 2023-08-24:
Not sure if anyone has tried this or if its even supported but was curious if there was any support for 2 or more waterfurnace devices on a single machine?

My 2 water furnace units are relatively close together so I am curious if its possible to pull data off both using one machine vs having 2

ccutrer, 2023-08-24:
Yes, that works just fine. You just need two RS485 adapters, and you create two service files, each one with a unique name and referencing the proper serial port. you may want to use udev rules to keep them straight, but it doesn't really matter if that's all you have on the box - the bridge will read the serial number from the connection and use that when publishing to MQTT.

starsoccer, 2023-08-24:
Okay great, waiting on my second adapter to arrive but then will give that a try.

Regarding the service files, I can tweak that. Do I just run one service though, `aurora_mqtt_bridge`, or do I need to make a second service and have 2 instances running?

Port wise, Its not a huge deal. It would be ideal if one service file could be used rather then 2 to also avoid duplicating mqtt connections which I assume is what will currently happen. I was originally going to use 2 different sets of creds, one for each unit but may no longer bother.

ccutrer, 2023-08-24:
Two service files running separate instances of the bridge, so two total MQTT connections. In theory I could run two in the same process, but it complicates several things, and I'd want to do two MQTT connections anyway since you can't have multiple last will and testament messages registered on a single connection. It's easy enough to just run separate services.

starsoccer, 2023-08-24:
Hmm okay interesting well good to know, Ill play around with getting 2 running then and may make a PR to add some extra documentation about it if I run into any issues