ccutrer/waterfurnace_aurora #48: Setup questions.

daviss57, 2023-09-07:
Hope you can help, these may be very basic questions as I am new to MQTT and not 100%

I am using Home Assistant on a RPI4 and have followed your instrucitons to setup and read from my Waterfurnace.

A few questions.
1. I have setup on a seperate RPI3 that I had laying around, I went through, when i typed sudo systemctl enable aurora_mqtt_bridge and sudo systemctl start aurora_mqtt_bridge the system seemed to pause for a few seconds but I did not get any response. Is this correct?
2. Every time i restart do i need to enable and start the service again?
3. Once i had it up and running on my RPI3 I was not 100% with regard to setting up in HA? Could you elaborate? i was not sure  what MQTT broker the service on the RPI3 was running.

I hope these are not too basic, appreciate any help getting this up and running.

Thank you in advance.

ccutrer, 2023-09-07:
1. Yes, but you can use `systemctl status aurora_mqtt_bridge` to see if it's actually running, or if it failed on startup.
2. No, `systemctl enable` sets it up to start at startup automatically
3. It should be running mosquitto. Note that HA can only connect to one MQTT broker at a time (at least last time I checked), so if you're running MQTT on your HA server (or elsewhere) you'll need to configure aurora_mqtt_bridge.service to connect to that broker, instead of localhost. Otherwise, you'll configure HA to connect to your Pi.

daviss57, 2023-09-08:
Thanks, I have Mosquitto up and running, do I have to setup my username and password on my PI that is connected to my waterfurnace? I dont recall setting it up.

ccutrer, 2023-09-17:
Mosquitto defaults to not requiring authentication.

daviss57, 2023-09-25:
Still having some issues geting setup.

What should my Broker be (se snip from HA)

[image]

daviss57, 2023-09-26:
Ok, some more basic questions as I seem to have confused myself.

So, in the setup it appears to install on a standalone Pi, I do not fully understand how the integration with Home Assistant works.

Is the intention here if running home assistant, to run two seperate Pi's?

I have currently set it up as two seperate Pi's is there an option to run everything on one?

If running two sperate Pi's then I just set up MQTT on my Home Assistant Server to connect to localhost which should connect to the Broker running on the second Pi once I have followed the setup as detailed in the readme file?

ccutrer, 2023-10-09:
I would say It depends on how things are physically available. I run the bridge, Mosquitto, and openHAB on the same NUC, because I have easy access to my heat pump from where my NUC is. If you need your Pi running Home Assistant in a separate physical location, that's fine. Or you can run it all on one Pi. You should just run a single instance of Mosquitto if you have multiple Pis, and then configure the bridge to connect to MQTT on the Home Assistant Pi, or vice versa.

closed 2023-10-09 by ccutrer