ccutrer/waterfurnace_aurora #61: ABC Connection Cable Pin Description Error

jim0020, 2024-03-10:
In the section "Connecting to the ABC", it says "Connect pins 1 and 3 (white/orange and white/green for a TIA-568-B configured cable) to + and pins 2 and 4 (orange and green)"

In TIA-568B, pin 4 is blue, not green.  So either the pin numbers given are wrong, or the colors are wrong.

[image]

0n3man, 2024-03-10:
If you look at the table right below that text it has orange and blue as being connected.  If you also look at the picture of the actual cable he built, you'll see it's orange and blue which match the pins numbers 2 and 4.  I built my cable off the picture of the cable, and it worked fine.

jim0020, 2024-03-10:
Thanks,  I found that section again with the picture.  There's a section below that that talks about using orange and green, which I kept finding first when scrolling up from the bottom and reading about the different commands.  I'm getting nothing over the connection, but I'm using a different RS485 to USB converter.  I ordered the one linked in the readme, which will be arriving tomorrow.

commit 482bde7e (2024-12-28): fix reference to wrong color in CAT5

closes #61
closed 2024-12-28 by ccutrer