ccutrer/waterfurnace_aurora #66: Compatibility with Aurora on Geostar?

minnehahak, 2024-07-23:
Wondering if anyone has gotten this package to work with a Geostar unit. 

Geostar is made by Waterfurnace as their slightly more budget brand, but has the exact same Aurora tech (uses an AID tool) and RJ45 AID tool port.

Originally I could not get MQTT Explorer to connect. I was on a Raspberry Pi 0W with 32-bit Raspberry Lite OS. I changed over to a Pi 3B+ with the same OS. Still not able to connect. I found another [issue here](https://github.com/ccutrer/waterfurnace_aurora/issues/27) and edited to the /etc/mosquitto/mosquitto.conf to contain the port and and allow anonymous. I was then able to connect to MQTT via the explorer. 

I'm getting a read out in MQTT that updates every few seconds, but it's full of only counts of "sent" "received" and other non useful information.
[image]

Thank you!

-----------
If helpful, here is the plate info on front of the unit:
[image]

And here is my serial number breakdown with the variables highlighted:
[image]

ccutrer, 2024-07-24:
I know of at least one person that is using it with GeoStat branded units.

minnehahak, 2024-07-24:
Got it, thank you. Is there somewhere you could point me to to find logs regarding what's going wrong? Thank you.

ccutrer, 2026-02-09:
You should try running it _not_ as a system service, so you can observe any errors connecting to MQTT.