ccutrer/waterfurnace_aurora #68: Readme Clarification

tejaycar, 2024-08-29:
First, thanks for the awesome work on this.  I'm still peeved that WF doesn't see fit to make this data available to consumers.  I can't wait to play this awesome code.

The readme states
"It is possible to directly connect to the GPIO on a Raspberry Pi"
Would it be possible to add some information on how to do that?  I assume it is as simple as connecting 1&3 and 2&4 to specific GPIO pins, right?

Second, the readme also mentions:
"The other pins are C and R from the thermostat bus, providing 24VAC power. DO NOT SHORT THESE PINS AGAINST ANYTHING, such as the communication pins, or a ground connection anywhere."

It seems those pins are totally unused.  Would it be possible, for those of us building our own cable, to simply not include those wires in the RJ-45 connector in the first place, thus ensuring we don't have any issues?  I assume yes, but I want to make sure I'm not missing something.

ccutrer, 2024-08-29:
> Would it be possible to add some information on how to do that? I assume it is as simple as connecting 1&3 and 2&4 to specific GPIO pins, right?

I don't want to include information on something that I've not done myself, nor had anyone conclusively show me can be done. But no, it's not simply connecting to the correct GPIOs, due to the need for level shifters and possibly other electronics. The one whisper I've heard of something doing it, they used an external RS485 chip (MAX485 I think?). But that comes with its own problem that it needs an additional signal sent when you're writing to the bus, and thus isn't a proper serial port to the operating system.

> Would it be possible, for those of us building our own cable, to simply not include those wires in the RJ-45 connector in the first place

Of course!

tejaycar, 2024-08-29:
Ah.  Ok.  That makes more sense.
Thank you!