ccutrer/waterfurnace_aurora #70: Error trying to start the aurora_mqtt_bridge service

ericmagaha, 2024-10-06:
Got everything set up on a RPI 3 and using the recommended USB 485 adapter.  I followed the "easy path" here -> https://github.com/ccutrer/waterfurnace_aurora?tab=readme-ov-file#getting-started-on-a-raspberry-pi

`eric@geothermal:/dev $ sudo systemctl status aurora_mqtt_bridge
● aurora_mqtt_bridge.service - Aurora MQTT Bridge
     Loaded: loaded (/etc/systemd/system/aurora_mqtt_bridge.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Sun 2024-10-06 17:02:11 EDT; 1s ago
    Process: 913 ExecStart=/usr/local/bin/aurora_mqtt_bridge /dev/ttyUSB0 mqtt://localhost/ (code=exited, stat>
   Main PID: 913 (code=exited, status=217/USER)
        CPU: 0`

If I try to run the command manually I get the following.
`eric@geothermal:~ $ /usr/local/bin/aurora_mqtt_bridge /dev/ttyUSB0 mqtt://localhost/
/var/lib/gems/3.1.0/gems/waterfurnace_aurora-1.4.8/lib/aurora/modbus/server.rb:79:in `<top (required)>': uninitialized constant ModBus::Server::Funcs (NameError)

ModBus::Server::Funcs.push(65, 66, 67, 68)
              ^^^^^^^
Did you mean?  ModBus::Server::FUNCS
	from <internal:/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb>:85:in `require'
	from <internal:/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb>:85:in `require'
	from /var/lib/gems/3.1.0/gems/waterfurnace_aurora-1.4.8/lib/aurora.rb:7:in `<top (required)>'
	from <internal:/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb>:85:in `require'
	from <internal:/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb>:85:in `require'
	from /var/lib/gems/3.1.0/gems/waterfurnace_aurora-1.4.8/exe/aurora_mqtt_bridge:4:in `<top (required)>'
	from /usr/local/bin/aurora_mqtt_bridge:25:in `load'
	from /usr/local/bin/aurora_mqtt_bridge:25:in `<main>'`

If I modify the server.rb to make the method uppercase I get the following.
`eric@geothermal:~ $ /usr/local/bin/aurora_mqtt_bridge /dev/ttyUSB0 mqtt://localhost/
/var/lib/gems/3.1.0/gems/waterfurnace_aurora-1.4.8/lib/aurora/modbus/server.rb:79:in `push': can't modify frozen Array: [1, 2, 3, 4, 5, 6, 15, 16, 22, 23] (FrozenError)
	from /var/lib/gems/3.1.0/gems/waterfurnace_aurora-1.4.8/lib/aurora/modbus/server.rb:79:in `<top (required)>'
	from <internal:/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb>:85:in `require'
	from <internal:/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb>:85:in `require'
	from /var/lib/gems/3.1.0/gems/waterfurnace_aurora-1.4.8/lib/aurora.rb:7:in `<top (required)>'
	from <internal:/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb>:85:in `require'
	from <internal:/usr/lib/ruby/vendor_ruby/rubygems/core_ext/kernel_require.rb>:85:in `require'
	from /var/lib/gems/3.1.0/gems/waterfurnace_aurora-1.4.8/exe/aurora_mqtt_bridge:4:in `<top (required)>'
	from /usr/local/bin/aurora_mqtt_bridge:25:in `load'
	from /usr/local/bin/aurora_mqtt_bridge:25:in `<main>'`

Help?  TIA.

ericmagaha, 2024-10-07:
Just noticed the documentation said install Ruby 2.5, 2.6 or 2.7. Ruby 3.0 has not been tested.
I installed Ruby 3.1.2. Guessing that is the issue. Will fix that tomorrow and try again.

ericmagaha, 2024-10-07:
ugh, same thing happens with Ruby 2.7.4.

ericmagaha, 2024-10-07:
Ok, got it working.  Cliff notes:

1.  Hardware is a Raspberry Pi model 3 that I had sitting around in a box.
2. Tried the latest Raspberry Pi OS based on Debian Bookworm which installs Ruby 3 by default.  That did not work as I got the Ruby error specified in the 1st post..
3. Tried Raspberry Pi OS based on Debian Bullseye which installs Ruby 2.7.4 by default.  That did not work, got the same Ruby error as specified in the 1st post.
4. Downloaded the next older Raspian based on Debian Buster.  That version installs Ruby 2.5 by default.  Ran into some install issues with the rmodbus-ccutrer which was trying to install 2.1.1 which said it needed Ruby 2.7.  Installed rmodbus-ccutrer 2.1.0.
5. Whatever version of the ffi gem it tried to install failed.  It was not compatible with the version of rubygems.  Installed ffi 1.9.3 which one of the other gems was dependent on (ccutrer serial or something iirc).
6. The systemd script /etc/systemd/system/aurora_mqtt_bridge.service specifies "pi" as the user.  Change that to an existent user account on your Raspberry Pi otherwise it does not work.

It is working and I can view the info using MQTT Explorer on my Macbook.

[image]

closed 2024-10-07 by ericmagaha

ccutrer, 2024-10-07:
v1.4.9 has been released, which should fix compatibility issues with both Ruby 2.5 and Ruby 3.3 (and everything in-between).

ericmagaha, 2024-10-07:
Ok.  I see you also released 1.4.10 and 1.4.11.  Should 1.4.11 be used?

I'll try it out when I get some time on a different SD card with the latest Raspberry Pi OS based on Debian Bookworm which installs Ruby 3 by default.