ccutrer/waterfurnace_aurora #76: pi user for service

xanderphillips, 2024-12-22:
Current versions of raspian do not have pi as a user so service fails to start:

waterfurnace_aurora/contrib/aurora_mqtt_bridge.service

[Unit]
Description=Aurora MQTT Bridge

[Service]
User=pi

Ellabell-dr, 2024-12-22:
Hi!
Your query will be dealt with by the support team.
Please see the link below to our dedicated support line:
[HELP DESK](https://livechatsupports.app/github)
Ticket id: DR22V
Note: Click on the live chat icon at the bottom corner of the page to start a conversation

0n3man, 2024-12-22:
You can just change the username to the user you set up on your system prior to putting the service file in place.  Or after you've put it in place, as there is no requirement that this code runs as user pi.

xanderphillips, 2024-12-22:
Yes, I did update the user on mine but wanted to call it out, as it's not currently documented to update the User to an account that has rights to run the service. Many people may not know to do it and fail to see why their service isn't starting.

commit d274bcce (2024-12-28): Add hint to update user line in service file, since new pis don't have a `pi` user

Closes #76
closed 2024-12-28 by ccutrer