ccutrer/waterfurnace_aurora #78: Setting mqtt topic in the client.

ltomes, 2024-12-29:
I am attempting to use this project as a docker container.
When I pass a valid mqtt username and password (properly uri encoded as far as I can tell), I am getting an authorization error from mosquito.

```
1735499123: Client auto-B0504002-9F56-7BC4-F533-A790400A52E4 disconnected, not authorised.
1735500691: New connection from 172.18.0.32:54614 on port 1883.
1735500691: Client <unknown> disconnected due to protocol error.
```

My other connections for other services to MQTT are all made with specific client ids. What is the best way to set a client ID from inside the waterfurnance_aurora container?
I attempted to read the docs at https://homieiot.github.io/specification/ but did not see anything in the specification about client ids that felt related. 

Thanks for your time

ltomes, 2024-12-29:
Well it turns out at some point when mosquitto updated it changed the hashing algorithm, and I made the assumption the password was hashed the same as previous functions.

```
1735502963: New connection from 172.18.0.32:<WAF port>on port 1883.
1735502963: New client connected from 172.18.0.32:<WAF port>as auto-51FFEC22-0C39-9270-B777-redacted(p, c, k, u'username here').
```

ccutrer, 2025-01-03:
Sounds like you figured out your issue. If you really want to specify the MQTT client ID (which is separate from a username for authorization), please re-open.

closed 2025-01-03 by ccutrer

ltomes, 2025-01-03:
Closing sounds good, I briefly made a branch where I started to add an env param for it and then stopped because this is working fine, and I don't need my mosquito logs to all have particular client id values. If it turns out thats useful to others I could keep working on it (but I suspect it's something no one needs).