ccutrer/waterfurnace_aurora #80: Zone 1 temperature consistent sudden drop when zone is heating

ericmagaha, 2025-01-14:
Hello.

I have a water furnace 5 system with 2 zones.  It is a new system that was installed in October.
I have been running this software for a few months to view additional data.  It works great.  I do not use home assistant or MQTT, I just fetch the data I want from whatever registers using aurora_fetch and then parse it and display it on my custom display.

I am fetching zone 1 temperature from register 31_007.  I am seeing consistent strange readings when the system is running and zone 1 is calling for heat i.e. h1.

Example:
Zone 1 heating target - 68°F

The zone 1 temperature will gradually increase, and then when it gets to 68.8, it will immediately drop to 68.0 and then work its way back up to 68.5 and then the zone will go into standby.  The stage 1 differential is 0.5.

This happens regardless of the target temperature i.e. if the target is 70°F, then the zone 1 temp will go to 70.8, then drop to 70.0, then work its way back up to 70.5 and then go to standby.

I do not see this behavior with Zone 2.

Is there some calculation that the waterfurnace is doing or some other register I should be pulling Zone 1 temperature from?

Thanks.

ericmagaha, 2025-01-14:
After staring at the data I see that the register Zone 1 Ambient Temperature (31007): 68.7°F is showing the same data as Entering Air (567): 68.7°F and Entering Air (740): 68.7°F.

And the values track each other identically.  I am guessing that there really is not a true Zone 1 Temperature.

Does anyone have any more information related to this?

0n3man, 2025-02-04:
I don't know if it's really relevant, but I have a 10 year old system.  I use this software with home assistant (HA).  I only have one zone.  When I started using this software with HA in the summer I noticed that the ambient temperature being reported out would actually rise 2 to 3 degrees, as the house cooled as detected on other sensors, before it would then start to track down.  The behavior actually had the house cooling 2 degrees below the set point.  Then this winter I notice that the WF would randomly drop what was report as ambient temperature.  In some cause it would drop by over 2 degrees.  Dropping over 2 degrees will cause aux heat to kick in.  As I was using HA to slightly very the heat when everyone left the house and during the night I was able to also create an automation to catch Aux Heat kicking in and stop it.  This [link](https://community.home-assistant.io/t/waterfurnace-control-via-modbus-bridge-with-smart-control-automations/693199/1) provides details on what I did with HA.

ericmagaha, 2025-02-04:
Ha, I actually found your thread a little over an hour ago by searching google for "waterfurnace ambient thermostat".  I read your post and there does seem similarities with the reported temperature suddenly changing, but mine is ALWAYS the same AND repeatable i.e. up/down 0.8 degrees.
I made a post on reddit with more detailed info of what the temperature does.
https://www.reddit.com/r/geothermal/comments/1i89c94/waterfurnace_intellizone2_zone_1_temperature/
Thanks for the reply!

0n3man, 2025-02-05:
What version thermostat do you have?  I have the CC-U02 version, which they don't sell any more.  The new one is a CC-U03 and I'm curious if you have that version.  I'll probably get mine replace, but I'd love to know if this software works with the CC-U03 before I get mine replace.

ericmagaha, 2025-02-05:
Good question.  Is it indicated somewhere on the thermostat?

0n3man, 2025-02-05:
You can tell by the look.  The CC-U03 is pictured [here](https://www.waterfurnace.com/residential/products/accessories/thermostats) and the CC-U02 [here](https://laureynsunited.com/waterfurnace-thermostats/).  The CC-U03 is  better looking.  The display has 90 degree angle corners, while the CC-U02 display has rounded corners.  If you have the CC-U03 and could provide the firmware version that would also be helpful.  On CC-U02 it displays the software version when you select menu -> settings.  The version is at the top of the display.

ericmagaha, 2025-02-05:
I definitely have the CC-U02.

The CC-U03 literature indicates that it works with the Symphony Home Comfort Platform so I do not see why it would not work with this software.

0n3man, 2025-02-05:
My concern would be they do a software update when installing the CC-U03, which moves around registers that are being read by this software.

ericmagaha, 2025-02-05:
I guess that is a concern if you are using Home Assistant with this, which I am not.  I am just using aurora_fetch to get the raw data and process it so it does not matter what the "name" of the register is.

ccutrer, 2025-02-05:
> My concern would be they do a software update when installing the CC-U03, which moves around registers that are being read by this software.

I think this is _highly_ unlikely. The Aurora Web Link works essentially identically to this software, so _moving_ a register would break existing AWLs. Especially the “basic” ones like thermostat ambient and target temperatures. They could add new registers, and you’d need a newer AWL to read those. You can actually see that happen in the register map - I originally mapped registers with a 7 series with IZ2, but then had to find some things for other users that have say a 5 series without IZ2. Then I go back and look at mine and see that indeed I have data in the same registers as the 5 series, it’s just not as detailed as I get from the IZ2 range of registers.

ericmagaha, 2025-02-21:
I attached a graph of the zone temperature history over 4 hours.  In this graph the zone 1 setpoint is 67 and the zone 2 setpoint is 65.  You can clearly see zone 1 sudden changes i.e. 67.0 -> 67.8 and vice versa.  Also, the algorithm NEVER allows the temperature to drop below the setpoint of 67.
Zone 2 behaves as a typical control algorithm should i.e. 0.5 below, on, 0.5 above, off.

[image]

ericmagaha, 2025-03-07:
Anyone else see zone 1 temperature behavior similar to what my system exhibits in the above graph?

It definitely has to be something in the Intellizone2 thermostat software.

Majiir, 2025-04-11:
I have two 7-series units, one with one zone and another with two zones. Both of them have a CC-U01 thermostat in the main zone, and the two-zone unit has an Intellizone thermostat in the second zone.

For both units, in Zone 1, I see the Entering Air Temp exactly matching the Zone 1 Ambient Temp. But the Zone 2 Ambient Temp reads a different value and seems to be accurately reporting the temperature in that room.

Based on the data, I think the Zone 1 Ambient Temp is actually correct, and it's the Entering Air Temp that's wrong. One of my units is in a cold basement, so it's easy to see it cooling down when it's inactive. The leaving air temp cools and reaches equilibrium with the basement, but the "entering air temp" stays elevated.