ccutrer/waterfurnace_aurora #82: DHW Water Temperature Sensor missing since v1.4.12

evil-dog, 2025-01-20:
Something happened in v1.4.12 that removed the DHW Water Temperature Sensor from Home Assistant.

I was previously running v1.4.11 (it seems) up until I updated to v1.5 last night (which broke quite a bit, see Issue #81 for that).

After finding v1.5 sevierly broken I reverted back to v1.4.12 which I thought was working, but I've now found that the DHW Water Temperature sensor is missing. So now I've reverted back to my original v1.4.11 (Docker tag: sha-1054832). This has made everything work again.

FYI, I'm using the docker image.

ccutrer, 2025-01-20:
In v1.5, this sensor has been removed in favor of being part of the DHW water heater component.

evil-dog, 2025-01-20:
Ah, didn't see that added.

Though now that sensor is not available for use without a bit of other work.

To be able to use the Current Temperature Attribute in cards you need to create a helper template sensor.

Settings > Devices and Services > Helpers > Create Helper > Template > Template a Sensor

The State Template should be something like:
`{{ states.water_heater.waterfurnace_domestic_hot_water_generator.attributes.current_temperature }}`

You'll also want to fill out the following fields:
Units: F
Device Class: Temperature
State Class: Measurement
Device: WaterFurnace