ccutrer/waterfurnace_aurora #83: Humidifier Component Unavailable

evil-dog, 2025-01-22:
Something is up with the Humidifier Component in v1.5.3.  This was working in the past was working in v1.4.11, though I do remember thinking something was odd there too, but it at least worked. It showed when the humidifier was running and the set point. Now Home Assistant shows the Humidifier as Unavailable. When I look in the history I see status from yesterday, but that was likely before I upgraded to v1.5.3.

I have a whole house humidifier connected to my WaterFurnace and triggered by the accessory relay.

I've confirmed that my dip switches are reporting correctly:
```
# ./aurora_fetch /dev/ttyUSB0 33
DIP Switch Status (33): {:fp1=>15, :fp2=>30, :reversing_valve=>:o, :accessory_relay=>:humidifier, :compressor=>2, :lockout=>:continuous, :dehumidifier_reheat=>:dehumidifier}
```

Thanks.

evil-dog, 2025-01-22:
Found this error message in my logs:

`
2025-01-22 00:52:23.372 ERROR (MainThread) [homeassistant.components.mqtt.entity] Error 'some but not all values in the same group of inclusion 'available_modes' @ data[<available_modes>]' when processing MQTT discovery message topic: 'homeassistant/humidifier/aurora-161002058/humidifier/config', message: '{'payload_off': 'false', 'payload_on': 'true', 'device_class': 'humidifier', 'command_topic': 'homie/aurora-161002058/humidistat/humidifier-running/set', 'state_topic': 'homie/aurora-161002058/humidistat/humidifier-running', 'target_humidity_state_topic': 'homie/aurora-161002058/humidistat/humidification-target', 'target_humidity_command_topic': 'homie/aurora-161002058/humidistat/humidification-target/set', 'target_humidity_command_template': '{{ value | round(0) }}', 'mode_state_topic': 'homie/aurora-161002058/humidistat/humidifier-mode', 'mode_command_topic': 'homie/aurora-161002058/humidistat/humidifier-mode/set', 'device': {'manufacturer': 'WaterFurnace', 'sw_version': '1.5.3', 'model': 'NDV049H111CTR0BA', 'identifiers': 'aurora-161002058', 'name': 'WaterFurnace'}, 'unique_id': 'aurora-161002058_humidifier', 'availability': [{'topic': 'homie/aurora-161002058/$state', 'payload_available': 'ready', 'payload_not_available': 'lost'}], 'qos': 1, 'name': 'Humidifier'}'
`

ccutrer, 2025-01-29:
This should be fixed in 1.5.4, but that error message was a little unclear, so I'm not 100% sure.

commit 2443030d (2025-01-29): update mqtt-homie-homeassistant and mqtt-homeassistant

fixes #83